pycom/pymakr-atom #249: Can't install package on Atom

mmr689, 2021-04-23:
> @serialport/bindings@8.0.8 install C:\Users\migue\AppData\Local\Temp\apm-install-dir-2021323-12388-n2bsa1.yrf3\node_modules\pymakr\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

C:\Users\migue\AppData\Local\Temp\apm-install-dir-2021323-12388-n2bsa1.yrf3\node_modules\pymakr\node_modules\@serialport\bindings>if not defined npm_config_node_gyp (node "C:\Users\migue\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Users\migue\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild ) 

npm WARN deprecated crypto@1.0.1: This package is no longer supported. It's now a built-in Node module. If you've depended on crypto, you should switch to the one that's built-in.
npm WARN deprecated parser-byte-length@1.0.2: renamed to @serialport/parser-byte-length
npm WARN deprecated parser-cctalk@1.0.2: reanmed to @serialport/parser-cctalk
npm WARN deprecated parser-readline@1.0.2: reanmed to @serialport/parser-readline
npm WARN deprecated parser-delimiter@1.0.2: reanmed to @serialport/parser-delimiter
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated parser-ready@1.0.2: reanmed to @serialport/parser-ready
npm WARN deprecated parser-regex@1.0.2: reanmed to @serialport/parser-regex
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
prebuild-install WARN install No prebuilt binaries found (target=9.4.4 runtime=electron arch=x64 libc= platform=win32)
gyp ERR! find VS 
gyp ERR! find VS msvs_version not set from command line or npm config
gyp ERR! find VS VCINSTALLDIR not set, not running in VS Command Prompt
gyp ERR! find VS could not use PowerShell to find Visual Studio 2017 or newer
gyp ERR! find VS looking for Visual Studio 2015
gyp ERR! find VS - not found
gyp ERR! find VS not looking for VS2013 as it is only supported up to Node.js 8
gyp ERR! find VS 
gyp ERR! find VS **************************************************************
gyp ERR! find VS You need to install the latest version of Visual Studio
gyp ERR! find VS including the "Desktop development with C++" workload.
gyp ERR! find VS For more information consult the documentation at:
gyp ERR! find VS https://github.com/nodejs/node-gyp#on-windows
gyp ERR! find VS **************************************************************
gyp ERR! find VS 
gyp ERR! configure error 
gyp ERR! stack Error: Could not find any Visual Studio installation to use
gyp ERR! stack     at VisualStudioFinder.fail (C:\Users\migue\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:121:47)
gyp ERR! stack     at C:\Users\migue\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:74:16
gyp ERR! stack     at VisualStudioFinder.findVisualStudio2013 (C:\Users\migue\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:351:14)
gyp ERR! stack     at C:\Users\migue\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:70:14
gyp ERR! stack     at C:\Users\migue\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:372:16
gyp ERR! stack     at C:\Users\migue\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\util.js:54:7
gyp ERR! stack     at C:\Users\migue\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\util.js:33:16
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:297:5)
gyp ERR! stack     at ChildProcess.emit (events.js:200:13)
gyp ERR! stack     at maybeClose (internal/child_process.js:1021:16)
gyp ERR! System Windows_NT 10.0.19041
gyp ERR! command "C:\\Users\\migue\\AppData\\Local\\atom\\app-1.56.0\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\migue\\AppData\\Local\\atom\\app-1.56.0\\resources\\app\\apm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\migue\AppData\Local\Temp\apm-install-dir-2021323-12388-n2bsa1.yrf3\node_modules\pymakr\node_modules\@serialport\bindings
gyp ERR! node -v v12.4.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm WARN notsup Unsupported engine for pymakr@2.1.12: wanted: {"atom":">=1.41.0","node":">=6.3.0 <=7.0.0"} (current: {"node":"12.4.0","npm":"6.14.8"})
npm WARN notsup Not compatible with your version of node/npm: pymakr@2.1.12
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\migue\AppData\Local\Temp\apm-install-dir-2021323-12388-n2bsa1.yrf3\package.json'
npm WARN apm-install-dir-2021323-12388-n2bsa1.yrf3 No description
npm WARN apm-install-dir-2021323-12388-n2bsa1.yrf3 No repository field.
npm WARN apm-install-dir-2021323-12388-n2bsa1.yrf3 No README data
npm WARN apm-install-dir-2021323-12388-n2bsa1.yrf3 No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@8.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings@8.0.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\migue\.atom\.apm\_logs\2021-04-23T06_58_09_280Z-debug.log

-----

In my VSC works good but I need it in Atom.

elmasria, 2021-04-23:
@mmr689 can you please provide us with the version of Atom, pymakr and which OS platform you are using (64, 32)

mmr689, 2021-04-23:
@elmasria 

Atom: 1.56.0
Pymark package 2.1.12
Win10 64

HvdWetering, 2021-04-29:
Have exactly same issue,
Atom: 1.56.0
Pymakr: 2.1.12
Win10 64
Same thing happens on different machines.

elmasria, 2021-04-30:
@HvdWetering and @mmr689 Ok I will take a look and try multiple environments (because It is working for me) but I will try on other pcs. Sorry for that will try to fix it soon

HvdWetering, 2021-04-30:
It happened on my desktop where an older version of Atom worked fine for a year or so, together with an older version of pymakr.
Atom pushed for an update, did it, also had to install latest pymakr, ans since then this problem is there.
Next on a laptop same thing happened.
Going back to earlyer versions would be a (temporary) possibility?

mmr689, 2021-04-30:
@HvdWetering Yes, same for me and yes if you install older versions you can temporally solve it. Another option is using VSC.

@elmasria I know more people with the same problem :(

HvdWetering, 2021-05-01:
Installed the old version of Atom which i used to use: 1.41.0 and there pymakr 2.1.12 installed without problems and runs ok for me.

elmasria, 2021-05-17:
Hi,

Thank you for reporting this. 

I have checked this on multiple environments. 

After the latest upgrade of Atom, the **serialport** package (and because we are still using version *8.0.x*) tries to rebuild based on the new electron version (used by Atom on version 1.57). As a result, **Pymakr** may throw issues on environments that miss some setup, like the error you mentioned. 

One solution is by doing the below setup for windows users to use `Pymakr.

Steps:
* Install Python 2.x.x (https://www.python.org/downloads/release/python-2718/). 
   If you already have python installed, make sure you add it to the environment variable.
* Download Microsoft Visual C++ (2017 or later). (https://stackoverflow.com/a/54136652) 
[image]

The following link will have two solutions, one I mentioned above and an alternative. But the (windows-build-tools) option did not work for me on two different Pcs.
https://stackoverflow.com/a/59882818

We will work on updating   **serialport** package to avoid this type of issue in the future.

elmasria, 2021-05-17:
Similar Issue https://github.com/pycom/pymakr-atom/issues/256

closed 2021-05-17 by elmasria