pycom/pymakr-atom #250: Can't install on atom

taubsix3, 2021-05-03:
I tried to install pymakr the version 2.1.12 on atom version 1.56.0 x64 over the package installer from atom but it don't work. 
The Error Code:

Installing “pymakr@2.1.12” failed.Hide output…

> @serialport/bindings@8.0.8 install C:\Users\T\AppData\Local\Temp\apm-install-dir-202143-3140-1ogi3i7.gph5\node_modules\pymakr\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

C:\Users\T\AppData\Local\Temp\apm-install-dir-202143-3140-1ogi3i7.gph5\node_modules\pymakr\node_modules\@serialport\bindings>if not defined npm_config_node_gyp (node "C:\Users\T\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Users\T\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild ) 

npm WARN deprecated crypto@1.0.1: This package is no longer supported. It's now a built-in Node module. If you've depended on crypto, you should switch to the one that's built-in.
npm WARN deprecated parser-delimiter@1.0.2: reanmed to @serialport/parser-delimiter
npm WARN deprecated parser-readline@1.0.2: reanmed to @serialport/parser-readline
npm WARN deprecated parser-ready@1.0.2: reanmed to @serialport/parser-ready
npm WARN deprecated parser-cctalk@1.0.2: reanmed to @serialport/parser-cctalk
npm WARN deprecated parser-byte-length@1.0.2: renamed to @serialport/parser-byte-length
npm WARN deprecated parser-regex@1.0.2: reanmed to @serialport/parser-regex
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
prebuild-install WARN install No prebuilt binaries found (target=9.4.4 runtime=electron arch=x64 libc= platform=win32)
gyp ERR! find VS 
gyp ERR! find VS msvs_version not set from command line or npm config
gyp ERR! find VS VCINSTALLDIR not set, not running in VS Command Prompt
gyp ERR! find VS could not use PowerShell to find Visual Studio 2017 or newer
gyp ERR! find VS looking for Visual Studio 2015
gyp ERR! find VS - not found
gyp ERR! find VS not looking for VS2013 as it is only supported up to Node.js 8
gyp ERR! find VS 
gyp ERR! find VS **************************************************************
gyp ERR! find VS You need to install the latest version of Visual Studio
gyp ERR! find VS including the "Desktop development with C++" workload.
gyp ERR! find VS For more information consult the documentation at:
gyp ERR! find VS https://github.com/nodejs/node-gyp#on-windows
gyp ERR! find VS **************************************************************
gyp ERR! find VS 
gyp ERR! configure error 
gyp ERR! stack Error: Could not find any Visual Studio installation to use
gyp ERR! stack     at VisualStudioFinder.fail (C:\Users\T\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:121:47)
gyp ERR! stack     at C:\Users\T\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:74:16
gyp ERR! stack     at VisualStudioFinder.findVisualStudio2013 (C:\Users\T\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:351:14)
gyp ERR! stack     at C:\Users\T\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:70:14
gyp ERR! stack     at C:\Users\T\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:372:16
gyp ERR! stack     at C:\Users\T\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\util.js:54:7
gyp ERR! stack     at C:\Users\T\AppData\Local\atom\app-1.56.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\util.js:33:16
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:297:5)
gyp ERR! stack     at ChildProcess.emit (events.js:200:13)
gyp ERR! stack     at maybeClose (internal/child_process.js:1021:16)
gyp ERR! System Windows_NT 10.0.19042
gyp ERR! command "C:\\Users\\T\\AppData\\Local\\atom\\app-1.56.0\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\T\\AppData\\Local\\atom\\app-1.56.0\\resources\\app\\apm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\T\AppData\Local\Temp\apm-install-dir-202143-3140-1ogi3i7.gph5\node_modules\pymakr\node_modules\@serialport\bindings
gyp ERR! node -v v12.4.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm WARN notsup Unsupported engine for pymakr@2.1.12: wanted: {"atom":">=1.41.0","node":">=6.3.0 <=7.0.0"} (current: {"node":"12.4.0","npm":"6.14.8"})
npm WARN notsup Not compatible with your version of node/npm: pymakr@2.1.12
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\T\AppData\Local\Temp\apm-install-dir-202143-3140-1ogi3i7.gph5\package.json'
npm WARN apm-install-dir-202143-3140-1ogi3i7.gph5 No description
npm WARN apm-install-dir-202143-3140-1ogi3i7.gph5 No repository field.
npm WARN apm-install-dir-202143-3140-1ogi3i7.gph5 No README data
npm WARN apm-install-dir-202143-3140-1ogi3i7.gph5 No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@8.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings@8.0.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\T\.atom\.apm\_logs\2021-05-03T16_54_19_357Z-debug.log

developpef, 2021-05-05:
Hello,

Got the same error! :s

elmasria, 2021-05-17:
Hi,

Thank you for reporting this. 

I have checked this on multiple environments. 

After the latest upgrade of Atom, the **serialport** package (and because we are still using version *8.0.x*) tries to rebuild based on the new electron version (used by Atom on version 1.57). As a result, **Pymakr** may throw issues on environments that miss some setup, like the error you mentioned. 

One solution is by doing the below setup for windows users to use `Pymakr.

Steps:
* Install Python 2.x.x (https://www.python.org/downloads/release/python-2718/). 
   If you already have python installed, make sure you add it to the environment variable.
* Download Microsoft Visual C++ (2017 or later). (https://stackoverflow.com/a/54136652) 
[image]

The following link will have two solutions, one I mentioned above and an alternative. But the (windows-build-tools) option did not work for me on two different Pcs.
https://stackoverflow.com/a/59882818

We will work on updating   **serialport** package to avoid this type of issue in the future.

elmasria, 2021-05-17:
Similar Issue https://github.com/pycom/pymakr-atom/issues/256

closed 2021-05-17 by elmasria