pycom/pymakr-atom #251: Package fails to install to Atom 1.56, MacOS Big Sur on Apple M1 silicon

brockcraft, 2021-05-04:
## What are the steps to reproduce this issue?
1. Add package to Atom from: https://atom.io/packages/pymakr
2. Install pymakr@2.1.12 package from Atom package installer

## What happens?
Install fails with error (below)

## What were you expecting to happen?
Clean install and REPL launch.

## Any logs, error output, etc?
https://gist.github.com/brockcraft/cc08cd90be753a2b696f50490fcf55d5

## Any other comments?

## What versions of software are you using?
**Operating system:**
MacOS Big Sure 11.2.3 on Apple M1 Silicon

**Atom version:**
1.56.0

**Pymakr version:**
2.1.12

developpef, 2021-05-05:
Hello,

Got the same error yesterday on Windows and Atom v1.56.

Please find a fix quickly!

elmasria, 2021-05-05:
@developpef , can you try to remove the atom file on the root level (.atom) and reinstall pymakr again. just to make sure you are working with Windows 10 64  (please check if any update is missing to install for window)

developpef, 2021-05-05:
I managed to make Pymakr v2.1.12 work with Atom v1.55.0

But latests versions don't seem capatible, as pointed in many other discussions from last few days.

elmasria, 2021-05-05:
@brockcraft, A new version is published. please let us know if this version solved the issue you mentioned.

brockcraft, 2021-05-05:
@elmasria Nope. Still fails. interestingly, if I launch the install from https://atom.io/packages/pymakr and click the install button in Atom, the progress indicator in the Install button is animated for about 30 seconds, then stops with no error.

If I search for the package in Atom and install from there, it fails. And this is with the wipy connected and ready to go on a serial port.

Log: https://gist.github.com/brockcraft/5adbb611791d603da82015f47569aa54

ClemensGruber, 2021-05-06:
Same here on Win10. Installation is not possible. Auto update from Atom ruins the working setup and going back needs time to have a working setup again. I have wasted too much time with pycom stuff. :-(

elmasria, 2021-05-17:
@ClemensGruber, please check Similar Issue https://github.com/pycom/pymakr-atom/issues/256

elmasria, 2021-05-17:
@brockcraft, please try to install `Xcode or CLT version` check the below.

https://medium.com/flawless-app-stories/gyp-no-xcode-or-clt-version-detected-macos-catalina-anansewaa-38b536389e8d

Loketony, 2021-06-23:
I've got the same error despite that I already have Xcode installed!

What worked for me was "retelling" xcode-select the path to Xcode:
`sudo xcode-select -s /Applications/Xcode.app/Contents/Developer`

closed 2021-06-24 by elmasria