pycom/pymakr-atom #256: Installing “pymakr@2.1.13” on ATOM failed.

ElectronicsInnovation, 2021-05-12:
> @serialport/bindings@8.0.8 install C:\Users\subra\AppData\Local\Temp\apm-install-dir-2021412-11740-146ez7j.qist\node_modules\pymakr\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

C:\Users\subra\AppData\Local\Temp\apm-install-dir-2021412-11740-146ez7j.qist\node_modules\pymakr\node_modules\@serialport\bindings>if not defined npm_config_node_gyp (node "C:\Users\subra\AppData\Local\atom\app-1.57.0\resources\app\apm\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Users\subra\AppData\Local\atom\app-1.57.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild ) 

npm WARN deprecated crypto@1.0.1: This package is no longer supported. It's now a built-in Node module. If you've depended on crypto, you should switch to the one that's built-in.
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated parser-byte-length@1.0.2: renamed to @serialport/parser-byte-length
npm WARN deprecated parser-cctalk@1.0.2: reanmed to @serialport/parser-cctalk
npm WARN deprecated parser-readline@1.0.2: reanmed to @serialport/parser-readline
npm WARN deprecated parser-delimiter@1.0.2: reanmed to @serialport/parser-delimiter
npm WARN deprecated parser-regex@1.0.2: reanmed to @serialport/parser-regex
npm WARN deprecated parser-ready@1.0.2: reanmed to @serialport/parser-ready
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
prebuild-install WARN install No prebuilt binaries found (target=9.4.4 runtime=electron arch=x64 libc= platform=win32)
gyp ERR! find Python 
gyp ERR! find Python Python is not set from command line or npm configuration
gyp ERR! find Python Python is not set from environment variable PYTHON
gyp ERR! find Python checking if "python" can be used
gyp ERR! find Python - "python" is not in PATH or produced an error
gyp ERR! find Python checking if "python2" can be used
gyp ERR! find Python - "python2" is not in PATH or produced an error
gyp ERR! find Python checking if "python3" can be used
gyp ERR! find Python - "python3" is not in PATH or produced an error
gyp ERR! find Python checking if the py launcher can be used to find Python 2
gyp ERR! find Python - "py.exe" is not in PATH or produced an error
gyp ERR! find Python checking if Python is C:\Python27\python.exe
gyp ERR! find Python - "C:\Python27\python.exe" could not be run
gyp ERR! find Python checking if Python is C:\Python37\python.exe
gyp ERR! find Python - "C:\Python37\python.exe" could not be run
gyp ERR! find Python 
gyp ERR! find Python **********************************************************
gyp ERR! find Python You need to install the latest version of Python.
gyp ERR! find Python Node-gyp should be able to find and use Python. If not,
gyp ERR! find Python you can try one of the following options:
gyp ERR! find Python - Use the switch --python="C:\Path\To\python.exe"
gyp ERR! find Python   (accepted by both node-gyp and npm)
gyp ERR! find Python - Set the environment variable PYTHON
gyp ERR! find Python - Set the npm configuration variable python:
gyp ERR! find Python   npm config set python "C:\Path\To\python.exe"
gyp ERR! find Python For more information consult the documentation at:
gyp ERR! find Python https://github.com/nodejs/node-gyp#installation
gyp ERR! find Python **********************************************************
gyp ERR! find Python 
gyp ERR! configure error 
gyp ERR! stack Error: Could not find any Python installation to use
gyp ERR! stack     at PythonFinder.fail (C:\Users\subra\AppData\Local\atom\app-1.57.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-python.js:307:47)
gyp ERR! stack     at PythonFinder.runChecks (C:\Users\subra\AppData\Local\atom\app-1.57.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-python.js:136:21)
gyp ERR! stack     at PythonFinder.<anonymous> (C:\Users\subra\AppData\Local\atom\app-1.57.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-python.js:225:16)
gyp ERR! stack     at PythonFinder.execFileCallback (C:\Users\subra\AppData\Local\atom\app-1.57.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-python.js:271:16)
gyp ERR! stack     at exithandler (child_process.js:302:5)
gyp ERR! stack     at ChildProcess.errorhandler (child_process.js:314:5)
gyp ERR! stack     at ChildProcess.emit (events.js:223:5)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:270:12)
gyp ERR! stack     at onErrorNT (internal/child_process.js:456:16)
gyp ERR! stack     at processTicksAndRejections (internal/process/task_queues.js:81:21)
gyp ERR! System Windows_NT 10.0.19041
gyp ERR! command "C:\\Users\\subra\\AppData\\Local\\atom\\app-1.57.0\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\subra\\AppData\\Local\\atom\\app-1.57.0\\resources\\app\\apm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\subra\AppData\Local\Temp\apm-install-dir-2021412-11740-146ez7j.qist\node_modules\pymakr\node_modules\@serialport\bindings
gyp ERR! node -v v12.14.1
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm WARN notsup Unsupported engine for pymakr@2.1.13: wanted: {"atom":">=1.41.0","node":">=6.3.0 <=7.0.0"} (current: {"node":"12.14.1","npm":"6.14.13"})
npm WARN notsup Not compatible with your version of node/npm: pymakr@2.1.13
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\subra\AppData\Local\Temp\apm-install-dir-2021412-11740-146ez7j.qist\package.json'
npm WARN apm-install-dir-2021412-11740-146ez7j.qist No description
npm WARN apm-install-dir-2021412-11740-146ez7j.qist No repository field.
npm WARN apm-install-dir-2021412-11740-146ez7j.qist No README data
npm WARN apm-install-dir-2021412-11740-146ez7j.qist No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@8.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings@8.0.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\subra\.atom\.apm\_logs\2021-05-12T07_24_56_232Z-debug.log

Xykon, 2021-05-13:
Can you please try to install the latest version of Python from https://www.python.org

elmasria, 2021-05-17:
Hi,

Thank you for reporting this. 

I have checked this on multiple environments. 

After the latest upgrade of Atom, the **serialport** package (and because we are still using version *8.0.x*) tries to rebuild based on the new electron version (used by Atom on version 1.57). As a result, **Pymakr** may throw issues on environments that miss some setup, like the error you mentioned. 

One solution is by doing the below setup for windows users to use `Pymakr.

Steps:
* Install Python 2.x.x (https://www.python.org/downloads/release/python-2718/). 
   If you already have python installed, make sure you add it to the environment variable.
* Download Microsoft Visual C++ (2017 or later). (https://stackoverflow.com/a/54136652) 
[image]

The following link will have two solutions, one I mentioned above and an alternative. But the (windows-build-tools) option did not work for me on two different Pcs.
https://stackoverflow.com/a/59882818

We will work on updating   **serialport** package to avoid this type of issue in the future.

closed 2021-05-19 by elmasria