pycom/pymakr-atom #258: Troubles connecting Lopy with Pymakr and Atom

iamlao, 2021-05-21:
Hello, 

I am not able anymore to connect to my Lopy using Pymakr and Atom. 
### Versions 
Pymakr : 2.1.13
Atom : 1.57.0
Pycom Expansion board : 2.1A
Lopy : 1.0r

When I connect the board to my computer, I got this : 
[image]

I tried to update the firmware of the board but I got this : 
[image]

I closed Atom before trying the update 

I tried rebooting the board by pressing the buttons for several seconds but nothing changed.

have you already had this ?

gijsio, 2021-05-21:
Hi,
This seems to be an issue related to Windows (it mentions 'Access refused' in the error). Perhaps there is an issue with the drivers or security settings on your computer? Are you able to use any other devices on COM ports? Let me know!

iamlao, 2021-05-21:
I can use an ESP32 on COM5