pycom/pymakr-atom #259: Can't install on Atom

AtomwontWork, 2021-05-26:
Trying to get a pygate and lopy4 to test Lora stuff, but the pymakr 2.1.13 isn't installing on Atom. The install button becomes striped, but just returns to the normal install button afterwards. Other plugins are installing and uninstalling fine, but even a full reboot didn't fix the pymakr install issue.

elmasria, 2021-05-27:
Hi, what OS you have?

please check (https://github.com/pycom/pymakr-atom/issues/256), (https://github.com/pycom/pymakr-atom/issues/248)

AtomwontWork, 2021-06-02:
I have done both 256 and 248s suggestions and still nothing. I am using Atom 1.57, most up to date python with environment variables, most recent vs build tools, along with reboots and windows updates.

It is still behaving in the way originally described.

ajoeleht, 2021-06-14:
Hi, it looks like 2.1.13 is not for Atom. Installation fails with errors referring to Visual Studio. Tried on two computers, one with earlier Atom installation and earlier working version of Pymakr, the second having fresh Atom and pymakr 2.1.13. Both fail with similar errors. By the way, both computers actually have VS.

prebuild-install WARN install No prebuilt binaries found (target=9.4.4 runtime=electron arch=x64 libc= platform=win32)
gyp ERR! find VS
gyp ERR! find VS msvs_version not set from command line or npm config
gyp ERR! find VS VCINSTALLDIR not set, not running in VS Command Prompt
gyp ERR! find VS could not use PowerShell to find Visual Studio 2017 or newer
gyp ERR! find VS looking for Visual Studio 2015
gyp ERR! find VS - not found
gyp ERR! find VS not looking for VS2013 as it is only supported up to Node.js 8
gyp ERR! find VS
gyp ERR! find VS **************************************************************
gyp ERR! find VS You need to install the latest version of Visual Studio
gyp ERR! find VS including the "Desktop development with C++" workload.
gyp ERR! find VS For more information consult the documentation at:
gyp ERR! find VS https://github.com/nodejs/node-gyp#on-windows
gyp ERR! find VS **************************************************************
gyp ERR! find VS
gyp ERR! configure error
gyp ERR! stack Error: Could not find any Visual Studio installation to use
gyp ERR! stack     at VisualStudioFinder.fail (C:\Users\argo\AppData\Local\atom\app-1.57.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:121:47)
gyp ERR! stack     at C:\Users\argo\AppData\Local\atom\app-1.57.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:74:16
gyp ERR! stack     at VisualStudioFinder.findVisualStudio2013 (C:\Users\argo\AppData\Local\atom\app-1.57.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:351:14)
gyp ERR! stack     at C:\Users\argo\AppData\Local\atom\app-1.57.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:70:14
gyp ERR! stack     at C:\Users\argo\AppData\Local\atom\app-1.57.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:372:16
gyp ERR! stack     at C:\Users\argo\AppData\Local\atom\app-1.57.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\util.js:54:7
gyp ERR! stack     at C:\Users\argo\AppData\Local\atom\app-1.57.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\util.js:33:16
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:302:5)
gyp ERR! stack     at ChildProcess.emit (events.js:223:5)
gyp ERR! stack     at maybeClose (internal/child_process.js:1021:16)
gyp ERR! System Windows_NT 6.1.7601
gyp ERR! command "C:\\Users\\argo\\AppData\\Local\\atom\\app-1.57.0\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\argo\\AppData\\Local\\atom\\app-1.57.0\\resources\\app\\apm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\argo\.atom\packages\pymakr-atom-2.1.12\node_modules\@serialport\bindings
gyp ERR! node -v v12.14.1
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.3.2 (node_modules\onchange\node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.3.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@8.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @serialport/bindings@8.0.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\argo\.atom\.apm\_logs\2021-06-14T11_18_32_876Z-debug.log

elmasria, 2021-06-15:
@ajoeleht thank you for your feedback, 

This error is mentioned in this issue (https://github.com/pycom/pymakr-atom/issues/256) and you can solve this by following the stopes men one in this comment (https://github.com/pycom/pymakr-atom/issues/256#issuecomment-842119652).

Based on the logs you submitted you have to go directly to step number 2 (Download Microsoft Visual C++ (2017 or later). (https://stackoverflow.com/a/54136652)) 

Have a nice day.

closed 2021-06-29 by elmasria