pycom/pymakr-atom #260: Unable to install

Stazium, 2021-06-22:
9026 warn notsup Unsupported engine for pymakr@2.1.13: wanted: {"atom":">=1.41.0","node":">=6.3.0 <=7.0.0"} (current: {"node":"12.14.1","npm":"6.14.13"})
9027 warn notsup Not compatible with your version of node/npm: pymakr@2.1.13
9028 verbose notsup Not compatible with your version of node/npm: pymakr@2.1.13
9028 verbose notsup Required: {"atom":">=1.41.0","node":">=6.3.0 <=7.0.0"}
9028 verbose notsup Actual:   {"npm":"6.14.13","node":"12.14.1"}

elmasria, 2021-06-24:
Hi @Stazium,

which OS are you using? can you provide more details?

Best Regards,
Ahmad El Masri

Stazium, 2021-06-28:
Hey Ahmad,
I am sorry I didn't see your message earlier.
I am using Windows 10. I found a fix by installing Visual Studio 2019  with
the C++/NodeJS packages, without really understanding why. Now it functions
perfectly with Atom v.1.57

Thank you for your answer anyway.
Best regards

elmasria, 2021-06-29:
Hello @Stazium,

Kindly have a look at this error we described why the installation of vs and python are a required steps in the current time 

https://github.com/pycom/pymakr-atom/issues/256

Best Regards,
Ahmad EL Masri

closed 2021-06-29 by elmasria