pycom/pymakr-atom #265: Error installing pymkr in new Atom

pmattia90, 2021-09-14:
Can't install pymkr in the last release of atom.

elmasria, 2021-09-15:
HI @pmattia90,

Thank you for reporting this. will take a look and release a new version.

Best Regards,
Ahmad El Masri

elmasria, 2021-09-15:
Hi @pmattia90, 

If you can share with us more details. 

Maybe will help if you take a look on this https://github.com/pycom/pymakr-atom/issues/256#issuecomment-842119652

Best Regards,
Ahmad EL Masri

pmattia90, 2021-09-15:
Same error

Ottieni Outlook per Android<https://aka.ms/ghei36>

Maybe will help if you take a look on this #256 (comment)<https://github.com/pycom/pymakr-atom/issues/256#issuecomment-842119652>

elmasria, 2021-09-15:
We presented the solution in the comment section if you can check by installing python and the required libraries. Thank you

monty34view, 2021-11-18:
Having similar problem re installation failure of pymker in atom.  Any progress ? 

Thanks

jakobrosenberg, 2021-11-18:
Thanks for the feedback @monty34view.

Can you provide us with logs or information that could help troubleshoot your issue?

monty34view, 2021-11-22:
Jakob

Thank you for responding.

I received a message in atom to update packages.   pymkr was one of the
packages to update.  The other packages updated without problems.   The
following is the error message

I am using atom version 1.58.0

Thanks

John

 @***@***.*** install
***@***.***\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

***@***.***\bindings>if
not defined npm_config_node_gyp (node
"C:\Users\jhou2636\AppData\Local\atom\app-1.58.0\resources\app\apm\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js"
rebuild )  else (node
"C:\Users\jhou2636\AppData\Local\atom\app-1.58.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js"
rebuild )

npm WARN deprecated ***@***.***: This package is no longer supported.
It's now a built-in Node module. If you've depended on crypto, you
should switch to the one that's built-in.
npm WARN deprecated ***@***.***: request has been deprecated, see
https://github.com/request/request/issues/3142
npm WARN deprecated ***@***.***: renamed to
@serialport/parser-byte-length
npm WARN deprecated ***@***.***: reanmed to @serialport/parser-cctalk
npm WARN deprecated ***@***.***: reanmed to
@serialport/parser-delimiter
npm WARN deprecated ***@***.***: reanmed to
@serialport/parser-readline
npm WARN deprecated ***@***.***: reanmed to @serialport/parser-regex
npm WARN deprecated ***@***.***: reanmed to @serialport/parser-ready
npm WARN deprecated ***@***.***: this library is no longer supported
prebuild-install WARN install No prebuilt binaries found (target=9.4.4
runtime=electron arch=x64 libc= platform=win32)
gyp ERR! find Python
gyp ERR! find Python Python is not set from command line or npm configuration
gyp ERR! find Python Python is not set from environment variable PYTHON
gyp ERR! find Python checking if "python" can be used
gyp ERR! find Python - "python" is not in PATH or produced an error
gyp ERR! find Python checking if "python2" can be used
gyp ERR! find Python - "python2" is not in PATH or produced an error
gyp ERR! find Python checking if "python3" can be used
gyp ERR! find Python - "python3" is not in PATH or produced an error
gyp ERR! find Python checking if the py launcher can be used to find Python 2
gyp ERR! find Python - "py.exe" is not in PATH or produced an error
gyp ERR! find Python checking if Python is C:\Python27\python.exe
gyp ERR! find Python - "C:\Python27\python.exe" could not be run
gyp ERR! find Python checking if Python is C:\Python37\python.exe
gyp ERR! find Python - "C:\Python37\python.exe" could not be run
gyp ERR! find Python
gyp ERR! find Python **********************************************************
gyp ERR! find Python You need to install the latest version of Python.
gyp ERR! find Python Node-gyp should be able to find and use Python. If not,
gyp ERR! find Python you can try one of the following options:
gyp ERR! find Python - Use the switch --python="C:\Path\To\python.exe"
gyp ERR! find Python   (accepted by both node-gyp and npm)
gyp ERR! find Python - Set the environment variable PYTHON
gyp ERR! find Python - Set the npm configuration variable python:
gyp ERR! find Python   npm config set python "C:\Path\To\python.exe"
gyp ERR! find Python For more information consult the documentation at:
gyp ERR! find Python https://github.com/nodejs/node-gyp#installation
gyp ERR! find Python **********************************************************
[... 26 lines cut ...]
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok

{"atom":">=1.41.0","node":">=6.3.0 <=7.0.0"} (current:
{"node":"12.14.1","npm":"6.14.13"})
npm WARN notsup Not compatible with your version of node/npm: ***@***.***
npm WARN enoent ENOENT: no such file or directory, open
'C:\Users\jhou2636\AppData\Local\Temp\apm-install-dir-20211023-6872-1bl8f8m.xa55\package.json'
npm WARN apm-install-dir-20211023-6872-1bl8f8m.xa55 No description
npm WARN apm-install-dir-20211023-6872-1bl8f8m.xa55 No repository field.
npm WARN apm-install-dir-20211023-6872-1bl8f8m.xa55 No README data
npm WARN apm-install-dir-20211023-6872-1bl8f8m.xa55 No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @***@***.*** install: `prebuild-install
--tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @***@***.*** install script.
npm ERR! This is probably not a problem with npm. There is likely
additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\jhou2636\.atom\.apm\_logs\2021-11-22T23_32_29_412Z-debug.log

jakobrosenberg, 2021-11-23:
@monty34view thanks for the log.

Could you try uninstall, restart Atom and reinstall the plugin?

If that doesn't work, please run  `atom -v` in your terminal and post the output as well as your OS and processor.

monty34view, 2021-11-24:
Jakob

I have uninstalled pymakr, but it won't reinstall.

atom-v outputs

Atom    : 1.58.0
Electron: 9.4.4
Chrome  : 83.0.4103.122
Node    : 12.14.1

Operating system is Windows 10 Enterprise

Intel(R) Core(TM) i7-6600U CPU @ 2.60GHz   2.80 GHz

16 G Ram

Thanks

John

jakobrosenberg, 2021-11-24:
Can you explain what happens when you try to reinstall, are you getting any errors or logs?

elmasria, 2021-11-24:
@monty34view 

please check this [#256 (comment)](https://github.com/pycom/pymakr-atom/issues/256#issuecomment-842119652)

Have a nice day

IsidurPaine, 2021-12-01:
Hi everybody,
I have a similar problem here.
Here's a complete log.

Can anyone explain what's happening?
Thank you in advance.

9049 warn notsup Unsupported engine for pymakr@2.2.0: wanted: {"atom":">=1.41.0","node":">=6.3.0 <=7.0.0"} (current: {"node":"12.14.1","npm":"6.14.13"})
9050 warn notsup Not compatible with your version of node/npm: pymakr@2.2.0
9051 verbose notsup Not compatible with your version of node/npm: pymakr@2.2.0
9051 verbose notsup Required: {"atom":">=1.41.0","node":">=6.3.0 <=7.0.0"}
9051 verbose notsup Actual:   {"npm":"6.14.13","node":"12.14.1"}
9052 warn enoent ENOENT: no such file or directory, open 'C:\Users\piero\AppData\Local\Temp\apm-install-dir-2021111-11940-upxcr2.qi7oe\package.json'
9053 verbose enoent This is related to npm not being able to find a file.
9054 warn apm-install-dir-2021111-11940-upxcr2.qi7oe No description
9055 warn apm-install-dir-2021111-11940-upxcr2.qi7oe No repository field.
9056 warn apm-install-dir-2021111-11940-upxcr2.qi7oe No README data
9057 warn apm-install-dir-2021111-11940-upxcr2.qi7oe No license field.
9058 verbose stack Error: @serialport/bindings@8.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
9058 verbose stack Exit status 1
9058 verbose stack     at EventEmitter.<anonymous> (C:\Users\piero\AppData\Local\atom\app-1.58.0\resources\app\apm\node_modules\npm\node_modules\npm-lifecycle\index.js:332:16)
9058 verbose stack     at EventEmitter.emit (events.js:223:5)
9058 verbose stack     at ChildProcess.<anonymous> (C:\Users\piero\AppData\Local\atom\app-1.58.0\resources\app\apm\node_modules\npm\node_modules\npm-lifecycle\lib\spawn.js:55:14)
9058 verbose stack     at ChildProcess.emit (events.js:223:5)
9058 verbose stack     at maybeClose (internal/child_process.js:1021:16)
9058 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
9059 verbose pkgid @serialport/bindings@8.0.8
9060 verbose cwd C:\Users\piero\AppData\Local\Temp\apm-install-dir-2021111-11940-upxcr2.qi7oe
9061 verbose Windows_NT 10.0.19043
9062 verbose argv "C:\\Users\\piero\\AppData\\Local\\atom\\app-1.58.0\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\piero\\AppData\\Local\\atom\\app-1.58.0\\resources\\app\\apm\\node_modules\\npm\\bin\\npm-cli.js" "--globalconfig" "C:\\Users\\piero\\.atom\\.apm\\.apmrc" "--userconfig" "C:\\Users\\piero\\.atom\\.apmrc" "install" "https://www.atom.io/api/packages/pymakr/versions/2.2.0/tarball" "--target=9.4.4" "--disturl=https://atom.io/download/electron" "--arch=x64" "--global-style"
9063 verbose node v12.14.1
9064 verbose npm  v6.14.13
9065 error code ELIFECYCLE
9066 error errno 1
9067 error @serialport/bindings@8.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
9067 error Exit status 1
9068 error Failed at the @serialport/bindings@8.0.8 install script.
9068 error This is probably not a problem with npm. There is likely additional logging output above.
9069 verbose exit [ 1, true ]

closed 2022-04-06 by jakobrosenberg