pycom/pymakr-atom #268: can not install pymakr via atom

Mustafalou, 2021-11-22:
installation failed, can't install pymakr into atom. And cannot git clone your repository :(

Mustafalou, 2021-11-22:
nvm ^^

jakobrosenberg, 2021-11-23:
Hi @Mustafalou.

Are you getting any errors while installing or cloning the repository?

Mustafalou, 2021-11-23:
I got this on atom :
Installing “pymakr@2.1.13” failed.Hide output…

> @serialport/bindings@8.0.8 install C:\Users\yilma\AppData\Local\Temp\apm-install-dir-20211023-12108-1a5h3hv.7vap\node_modules\pymakr\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

C:\Users\yilma\AppData\Local\Temp\apm-install-dir-20211023-12108-1a5h3hv.7vap\node_modules\pymakr\node_modules\@serialport\bindings>if not defined npm_config_node_gyp (node "C:\Users\yilma\AppData\Local\atom\app-1.58.0\resources\app\apm\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Users\yilma\AppData\Local\atom\app-1.58.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild ) 

npm WARN deprecated crypto@1.0.1: This package is no longer supported. It's now a built-in Node module. If you've depended on crypto, you should switch to the one that's built-in.
npm WARN deprecated parser-byte-length@1.0.2: renamed to @serialport/parser-byte-length
npm WARN deprecated parser-regex@1.0.2: reanmed to @serialport/parser-regex
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated parser-cctalk@1.0.2: reanmed to @serialport/parser-cctalk
npm WARN deprecated parser-delimiter@1.0.2: reanmed to @serialport/parser-delimiter
npm WARN deprecated parser-readline@1.0.2: reanmed to @serialport/parser-readline
npm WARN deprecated parser-ready@1.0.2: reanmed to @serialport/parser-ready
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
prebuild-install WARN install No prebuilt binaries found (target=9.4.4 runtime=electron arch=x64 libc= platform=win32)
gyp ERR! find VS 
gyp ERR! find VS msvs_version not set from command line or npm config
gyp ERR! find VS VCINSTALLDIR not set, not running in VS Command Prompt
gyp ERR! find VS checking VS2019 (16.11.31911.196) found at:
gyp ERR! find VS "C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools"
gyp ERR! find VS - "Visual Studio C++ core features" missing
gyp ERR! find VS could not find a version of Visual Studio 2017 or newer to use
gyp ERR! find VS looking for Visual Studio 2015
gyp ERR! find VS - not found
gyp ERR! find VS not looking for VS2013 as it is only supported up to Node.js 8
gyp ERR! find VS 
gyp ERR! find VS **************************************************************
gyp ERR! find VS You need to install the latest version of Visual Studio
gyp ERR! find VS including the "Desktop development with C++" workload.
gyp ERR! find VS For more information consult the documentation at:
gyp ERR! find VS https://github.com/nodejs/node-gyp#on-windows
gyp ERR! find VS **************************************************************
gyp ERR! find VS 
gyp ERR! configure error 
gyp ERR! stack Error: Could not find any Visual Studio installation to use
gyp ERR! stack     at VisualStudioFinder.fail (C:\Users\yilma\AppData\Local\atom\app-1.58.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:121:47)
gyp ERR! stack     at C:\Users\yilma\AppData\Local\atom\app-1.58.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:74:16
gyp ERR! stack     at VisualStudioFinder.findVisualStudio2013 (C:\Users\yilma\AppData\Local\atom\app-1.58.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:351:14)
gyp ERR! stack     at C:\Users\yilma\AppData\Local\atom\app-1.58.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:70:14
gyp ERR! stack     at C:\Users\yilma\AppData\Local\atom\app-1.58.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:372:16
gyp ERR! stack     at C:\Users\yilma\AppData\Local\atom\app-1.58.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\util.js:54:7
gyp ERR! stack     at C:\Users\yilma\AppData\Local\atom\app-1.58.0\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\util.js:33:16
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:302:5)
gyp ERR! stack     at ChildProcess.emit (events.js:223:5)
gyp ERR! stack     at maybeClose (internal/child_process.js:1021:16)
gyp ERR! System Windows_NT 10.0.19042
gyp ERR! command "C:\\Users\\yilma\\AppData\\Local\\atom\\app-1.58.0\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\yilma\\AppData\\Local\\atom\\app-1.58.0\\resources\\app\\apm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\yilma\AppData\Local\Temp\apm-install-dir-20211023-12108-1a5h3hv.7vap\node_modules\pymakr\node_modules\@serialport\bindings
gyp ERR! node -v v12.14.1
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm WARN notsup Unsupported engine for pymakr@2.1.13: wanted: {"atom":">=1.41.0","node":">=6.3.0 <=7.0.0"} (current: {"node":"12.14.1","npm":"6.14.13"})
npm WARN notsup Not compatible with your version of node/npm: pymakr@2.1.13
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\yilma\AppData\Local\Temp\apm-install-dir-20211023-12108-1a5h3hv.7vap\package.json'
npm WARN apm-install-dir-20211023-12108-1a5h3hv.7vap No description
[... cut ...]
npm WARN apm-install-dir-20211023-12108-1a5h3hv.7vap No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@8.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings@8.0.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\yilma\.atom\.apm\_l

Mustafalou, 2021-11-23:
yesterday, this does not show 
[image]
but today, it is so i think cloning should have resolved it :)

jakobrosenberg, 2021-11-23:
Happy to hear you got it working. I'll close the issue for now, but feel free to reopen it if the issue returns.

closed 2021-11-23 by jakobrosenberg