pycom/pymakr-atom #271: Uncaught TypeError: _this.pymakr.sync is not a function

Hofkey, 2021-12-22:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What are the steps to reproduce this issue?
1. Try to sync a project with pymakr

## What happens?
An error occured, which has been reported many times as I can see.

## What were you expecting to happen?
That the project synchronization would actually work.

**Atom**: 1.58.0 x64
**Electron**: 9.4.4
**OS**: Microsoft Windows 10 Pro
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 2.2.0

### Stack Trace

Uncaught TypeError: _this.pymakr.sync is not a function

```
At C:\Users\beren\.atom\packages\pymakr\lib\main.js:45

TypeError: _this.pymakr.sync is not a function
    at HTMLElement.pymakrSync (/packages/pymakr/lib/main.js:45:47)
    at CommandRegistry.handleCommandEvent (~/AppData/Local/atom/app-1.58.0/resources/app/static/<embedded>:11:352326)
    at CommandRegistry.dispatch (~/AppData/Local/atom/app-1.58.0/resources/app/static/<embedded>:11:350801)
    at AtomEnvironment.dispatchApplicationMenuCommand (~/AppData/Local/atom/app-1.58.0/resources/app/static/<embedded>:1:833412)
    at EventEmitter.t (~/AppData/Local/atom/app-1.58.0/resources/app/static/<embedded>:1:841383)
    at EventEmitter.emit (events.js:223:5)
    at Object.onMessage (electron/js2c/renderer_init.js:115:818)
```

### Commands

```
     -0:02.8.0 pymakr:sync (div.tool-panel.tree-view)
```

### Non-Core Packages

```
pymakr 2.2.0 
```

jakobrosenberg, 2022-01-03:
Hi @Hofkey. I can't find any documentation for what the `synchronize project` is supposed to do, but I can confirm that the function doesn't exist. It was removed with this refactor https://github.com/pycom/pymakr-atom/commit/d257e3466f6c9ad4cbe6b011aa1eee990bc816c1#diff-e50bac1e5d6d8fce7f69e2187d110cac5f5c9176b01496337db7ddc50fc2e7f6.

I'll get back to you with an update once I know more.

jakobrosenberg, 2022-01-31:
:tada: This issue has been resolved in version 2.2.2 :tada:

The release is available on:
- [GitHub release](https://github.com/pycom/pymakr-atom/releases/tag/v2.2.2)
- [Atom package](https://atom.io/packages/pymakr)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

PR #286 (2022-01-31): Fix-removed-deprecated-sync-project
This PR gets rid of the deprecated "Packages->Pymakr->synchronize project".

sync project was deprecated in https://github.com/pycom/pymakr-atom/commit/d257e3466f6c9ad4cbe6b011aa1eee990bc816c1#diff-e50bac1e5d6d8fce7f69e2187d110cac5f5c9176b01496337db7ddc50fc2e7f6

closes #271
closed 2022-01-31 by jakobrosenberg