pycom/pymakr-atom #275: Atom doesn't detect my fipy

Clem-a-la-gome, 2022-01-07:
Hi !! I'm really new with pycom and I don't know how to decribe my problem with technical words so I'll try to tell you what happenned:

1. I installed pymakr
2. I connect for the first time a pycom device (fipy) with the expansion board 2.0 and a LoRa antenna
3. COM3 appeard but I couldn't write anything into the terminal
3. I wrote the following file named: my_first_file.py :

import pycom
pycom.heartbeat(False)
pycom.rgbled(0330033)

4. I upploaded the file into the COM3 portal
5. It didn't do anything, I look my code and I forgot a "x" beetwen the first 0 and the 3, so I did ctrl+alt c
6. The connect/disconnect button turned red
7. I couldn't reconnect the fipy with atom so I decided to disconnect it from the usb port
8. Atom doesn't recognize anything

[image]

So, how can I reconnect my Pycom device?

**Operating system:** windows 11

**Atom version:** last version

**Pymakr version:** last version

Clem-a-la-gome, 2022-01-07:
I've connected a Pygate with a LoPy device and atom detect the device.

Clem-a-la-gome, 2022-01-07:
I've switch beetween my LoPy and FiPy in the expansion board and it don't work on both configuration.

Clem-a-la-gome, 2022-01-07:
Some little details, I have an expansion board 2 and I don't have put the SD card (maybe here is the problem) yet.

Clem-a-la-gome, 2022-01-08:
I have try to put the SD card and atom doesn't recognize my device.

jakobrosenberg, 2022-01-10:
Hi @Clem-a-la-gome could you try PyMakr for VSCode instead? Sorry about the hassle.

Clem-a-la-gome, 2022-01-10:
I tried and it didn't work !

And I connect my pygate and it works well on both.

[image]

Clem-a-la-gome, 2022-01-10:
I don't know if it's the expansion board that have a problem. The led from my FiPy is blue.

[image]