pycom/pymakr-atom #297: Won't install in Atom

ilium007, 2022-04-01:
Won't install in Atom (current download, OSX)

```
> @serialport/bindings-cpp@10.7.0 install /private/var/folders/00/gtf1vnjd4rbcddth6b7dspf80000gn/T/apm-install-dir-202231-68808-1l67a26.q12q/node_modules/pymakr/node_modules/@serialport/bindings-cpp
> node-gyp-build

  CXX(target) Release/obj.target/bindings/src/serialport.o
  CXX(target) Release/obj.target/bindings/src/serialport_unix.o
  CXX(target) Release/obj.target/bindings/src/poller.o

npm WARN deprecated crypto@1.0.1: This package is no longer supported. It's now a built-in Node module. If you've depended on crypto, you should switch to the one that's built-in.
npm WARN deprecated parser-byte-length@1.0.2: renamed to @serialport/parser-byte-length
npm WARN deprecated parser-cctalk@1.0.2: reanmed to @serialport/parser-cctalk
npm WARN deprecated parser-delimiter@1.0.2: reanmed to @serialport/parser-delimiter
npm WARN deprecated parser-readline@1.0.2: reanmed to @serialport/parser-readline
npm WARN deprecated parser-regex@1.0.2: reanmed to @serialport/parser-regex
npm WARN deprecated parser-ready@1.0.2: reanmed to @serialport/parser-ready
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
../src/poller.cpp:108:7: error: no member named 'SetInstanceData' in 'Napi::Env'
  env.SetInstanceData<Napi::FunctionReference>(constructor);
  ~~~ ^
../src/poller.cpp:108:46: error: expected '(' for function-style cast or type construction
  env.SetInstanceData<Napi::FunctionReference>(constructor);
                      ~~~~~~~~~~~~~~~~~~~~~~~^
../src/poller.cpp:127:53: error: no member named 'GetInstanceData' in 'Napi::Env'
  Napi::FunctionReference* constructor = info.Env().GetInstanceData<Napi::FunctionReference>();
                                         ~~~~~~~~~~ ^
../src/poller.cpp:127:92: error: expected '(' for function-style cast or type construction
  Napi::FunctionReference* constructor = info.Env().GetInstanceData<Napi::FunctionReference>();
                                                                    ~~~~~~~~~~~~~~~~~~~~~~~^
../src/poller.cpp:127:94: error: expected expression
  Napi::FunctionReference* constructor = info.Env().GetInstanceData<Napi::FunctionReference>();
                                                                                             ^
5 errors generated.
make: *** [Release/obj.target/bindings/src/poller.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/private/var/folders/00/gtf1vnjd4rbcddth6b7dspf80000gn/T/apm-install-dir-202231-68808-1l67a26.q12q/node_modules/pymakr/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:223:5)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Darwin 21.2.0
gyp ERR! command "/Applications/Atom.app/Contents/Resources/app/apm/bin/node" "/private/var/folders/00/gtf1vnjd4rbcddth6b7dspf80000gn/T/apm-install-dir-202231-68808-1l67a26.q12q/node_modules/pymakr/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /private/var/folders/00/gtf1vnjd4rbcddth6b7dspf80000gn/T/apm-install-dir-202231-68808-1l67a26.q12q/node_modules/pymakr/node_modules/@serialport/bindings-cpp
gyp ERR! node -v v12.14.1
gyp ERR! node-gyp -v v6.1.0
gyp ERR! not ok 
npm WARN notsup Unsupported engine for @serialport/bindings-cpp@10.7.0: wanted: {"node":">=12.17.0 <13.0 || >=14.0.0"} (current: {"node":"12.14.1","npm":"6.14.13"})
npm WARN notsup Not compatible with your version of node/npm: @serialport/bindings-cpp@10.7.0
npm WARN notsup Unsupported engine for @serialport/bindings-interface@1.2.1: wanted: {"node":"^12.22 || ^14.13 || >=16"} (current: {"node":"12.14.1","npm":"6.14.13"})
npm WARN notsup Not compatible with your version of node/npm: @serialport/bindings-interface@1.2.1
npm WARN enoent ENOENT: no such file or directory, open '/private/var/folders/00/gtf1vnjd4rbcddth6b7dspf80000gn/T/apm-install-dir-202231-68808-1l67a26.q12q/package.json'
npm WARN apm-install-dir-202231-68808-1l67a26.q12q No description
npm WARN apm-install-dir-202231-68808-1l67a26.q12q No repository field.
npm WARN apm-install-dir-202231-68808-1l67a26.q12q No README data
npm WARN apm-install-dir-202231-68808-1l67a26.q12q No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings-cpp@10.7.0 install: `node-gyp-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings-cpp@10.7.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```

ilium007, 2022-04-02:
I can install serialport/bindings-cpp from the command prompt:

```
❯ npm i @serialport/bindings-cpp@10.7.0                                                                                                                                       
up to date, audited 83 packages in 1s

22 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities

```

ilium007, 2022-04-07:
@jakobrosenberg - whilst it now installs the USB serial devices are not present in the drop down list although I have two connected devices:

[image]

ilium007, 2022-04-07:
```
~/.atom/packages/pymakr                                                                                                                                                                                                                                                                                                      11:14:05
❯ apm install                                                                                                                                                                                                                                                                                              100%   LIBERTY2 104 Mbps
Installing modules ✗
> fsevents@1.2.13 install /Users/brantwinter/.atom/packages/pymakr/node_modules/fsevents
> node install.js

  SOLINK_MODULE(target) Release/.node
  CXX(target) Release/obj.target/fse/fsevents.o
  SOLINK_MODULE(target) Release/fse.node

> @serialport/bindings-cpp@10.7.0 install /Users/brantwinter/.atom/packages/pymakr/node_modules/@serialport/bindings-cpp
> node-gyp-build

  CXX(target) Release/obj.target/bindings/src/serialport.o
  CXX(target) Release/obj.target/bindings/src/serialport_unix.o
  CXX(target) Release/obj.target/bindings/src/poller.o

npm WARN read-shrinkwrap This version of npm is compatible with lockfileVersion@1, but package-lock.json was generated for lockfileVersion@2. I'll try to do my best with it!
../src/poller.cpp:108:7: error: no member named 'SetInstanceData' in 'Napi::Env'
  env.SetInstanceData<Napi::FunctionReference>(constructor);
  ~~~ ^
../src/poller.cpp:108:46: error: expected '(' for function-style cast or type construction
  env.SetInstanceData<Napi::FunctionReference>(constructor);
                      ~~~~~~~~~~~~~~~~~~~~~~~^
../src/poller.cpp:127:53: error: no member named 'GetInstanceData' in 'Napi::Env'
  Napi::FunctionReference* constructor = info.Env().GetInstanceData<Napi::FunctionReference>();
                                         ~~~~~~~~~~ ^
../src/poller.cpp:127:92: error: expected '(' for function-style cast or type construction
  Napi::FunctionReference* constructor = info.Env().GetInstanceData<Napi::FunctionReference>();
                                                                    ~~~~~~~~~~~~~~~~~~~~~~~^
../src/poller.cpp:127:94: error: expected expression
  Napi::FunctionReference* constructor = info.Env().GetInstanceData<Napi::FunctionReference>();
                                                                                             ^
5 errors generated.
make: *** [Release/obj.target/bindings/src/poller.o] Error 1
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/Users/brantwinter/.atom/packages/pymakr/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:223:5)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Darwin 21.4.0
gyp ERR! command "/Applications/Atom.app/Contents/Resources/app/apm/bin/node" "/Users/brantwinter/.atom/packages/pymakr/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/brantwinter/.atom/packages/pymakr/node_modules/@serialport/bindings-cpp
gyp ERR! node -v v12.14.1
gyp ERR! node-gyp -v v6.1.0
gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings-cpp@10.7.0 install: `node-gyp-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @serialport/bindings-cpp@10.7.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/brantwinter/.atom/.apm/_logs/2022-04-07T01_14_42_430Z-debug.log

~/.atom/packages/pymakr                                                                                                                                                                                                                                                                                                  29s 11:14:42
❯
```

ilium007, 2022-04-07:
Debug log:

[2022-04-07T01_17_25_160Z-debug.log](https://github.com/pycom/pymakr-atom/files/8438880/2022-04-07T01_17_25_160Z-debug.log)

ilium007, 2022-04-07:
Then tried to remove and reinstall:

[image]

ilium007, 2022-04-07:
My NPM is installed via homebrew ie. in /usr/local/bin/npm

```
❯ which npm
/usr/local/bin/npm
```

jakobrosenberg, 2022-04-07:
Thanks @ilium007, I'll have a look into this.

As a temporary solution, you can cd into pymakr and run `npm install serialport` manually. It should be installed in `~/.atom/packages/pymakr` if I remember correctly.

ilium007, 2022-04-07:
I tried that, made no difference

jakobrosenberg, 2022-04-07:
Can you check which version of `node` you have installed? (`node -v`)

ilium007, 2022-04-07:
```
❯ node -v 
v17.8.0
```

pgaufillet, 2022-04-13:
Hi,

Pymkr 2.3.7 + Atom 1.60.0 doesn't work on macOS 12.3.1.
Manual install of serialport in ~/.atom/packages/pymkr doesn't work (using Atom provided node/npm). Log:
```
191 warn notsup Unsupported engine for @serialport/bindings-cpp@10.7.0: wanted: {"node":">=12.17.0 <13.0 || >=14.0.0"} (current: {"node":"12.14.1","npm":"6.14.13"})
192 warn notsup Not compatible with your version of node/npm: @serialport/bindings-cpp@10.7.0
193 verbose notsup Not compatible with your version of node/npm: @serialport/bindings-cpp@10.7.0
193 verbose notsup Required: {"node":">=12.17.0 <13.0 || >=14.0.0"}
193 verbose notsup Actual:   {"npm":"6.14.13","node":"12.14.1"}
194 warn notsup Unsupported engine for @serialport/bindings-interface@1.2.1: wanted: {"node":"^12.22 || ^14.13 || >=16"} (current: {"node":"12.14.1","npm":"6.14.13"})
195 warn notsup Not compatible with your version of node/npm: @serialport/bindings-interface@1.2.1
196 verbose notsup Not compatible with your version of node/npm: @serialport/bindings-interface@1.2.1
196 verbose notsup Required: {"node":"^12.22 || ^14.13 || >=16"}
196 verbose notsup Actual:   {"npm":"6.14.13","node":"12.14.1"}
```