pycom/pymakr-atom #312: Does not detect UM Feather-S3

DandrewsDev, 2022-04-07:
## What are the steps to reproduce this issue?
1. Open atom with board already attached
2. Click connect device

## What happens?
There are no devices listed even though on is connected and working. 

## What were you expecting to happen?
The device to be listed and selectable.

## Any logs, error output, etc?
"No devices detected on USB"

## Any other comments?
The device is connected and I can get a repl connection via terminal. 

/dev/cu.usbmodem_ts3_1
Connecting via screen -port /dev/cu.usbmodem_ts3_1 115200 
Works great.

## What versions of software are you using?

**Operating system:**
Mac OS 12.3

**Atom version:**
1.60.0

**Pymakr version:**
2.3.7

jakobrosenberg, 2022-04-07:
Hi @DandrewsDev, you will have to update the plugin config to accommodate your device.

[image]

DandrewsDev, 2022-04-07:
Adding Unexpected Maker to the list of manufactures fixed it. I should have seen that config. Thank you.

closed 2022-04-07 by DandrewsDev