pycom/pymakr-atom #313: Pymakr 3.3.7 installer fails

Ridgeglider, 2022-04-24:
Installing “pymakr@2.3.7” failed.Hide output…

npm ERR! code E500
npm ERR! 500 Internal Server Error - GET https://www.atom.io/api/packages/pymakr/versions/2.3.7/tarball

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\HShar\.atom\.apm\_logs\2022-04-24T23_07_54_548Z-debug.log

Ridgeglider, 2022-04-24:
[2022-04-24T23_16_51_404Z-debug.log](https://github.com/pycom/pymakr-atom/files/8550702/2022-04-24T23_16_51_404Z-debug.log)
.

Ridgeglider, 2022-04-24:
here's the debug log..

jakobrosenberg, 2022-04-25:
Hi @Ridgeglider . Did you solve this issue?

Ridgeglider, 2022-04-25:
No. Issue is still open. Not sure how I managed to mark it as closed. Thanks in advance for any help.

elmasria, 2022-04-27:
@Ridgeglider I think this is most probably related to npm. can you try to install again

Ridgeglider, 2022-04-27:
I've tried several different installs. First the default just within Atom's 'install package', next run as admin, next run as admin w/ virus/firewalls off. Uninstall between attempts All fail.
Just tried it one more time with almost the same results except the 
 "**http fetch GET 500**" errors seem to be " **http fetch GET 400**" errors as shown in the attached .log file

[2022-04-27T15_45_43_389Z-debug.log](https://github.com/pycom/pymakr-atom/files/8574075/2022-04-27T15_45_43_389Z-debug.log)

The REPL console seems to be installed...?

jakobrosenberg, 2022-04-28:
Looking at the logs, it seems the error relates to this package: https://atom.io/packages/kite

```
19 error code E400
20 error 400 Bad Request - GET https://www.atom.io/api/packages/kite/versions/0.206.0/tarball
```

jakobrosenberg, 2022-05-02:
@Ridgeglider I'm using the latest version of Pymakr (2.3.7) with the latest version of Atom (1.60.0).

If you're comfortable with vscode, I'd recommend trying this: https://marketplace.visualstudio.com/items?itemName=pycom.pymakr-preview. To access the readmes, you'll have to visit the `next` branch. https://github.com/pycom/pymakr-vsc/tree/next