pycom/pymakr-atom #315: Failed to activate pymakr package Mac OS 12.3.1 Atom 1.60.0 x64

RaspChris, 2022-04-26:
Command failed: npm install serialport
/bin/sh: npm: command not found
  at checkExecSyncError (child_process.js:630:11)
    at Object.execSync (child_process.js:666:15)
    at Object.execSync (electron/js2c/asar.js:727:23)
    at Object.activate (/Users/christine/.atom/packages/pymakr-atom-develop/lib/main.js:17:34)
    at Package.activateNow (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:3412128)
    at /Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:3411540
    at Package.measure (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:3409118)
    at /Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:3411400
    at new Promise (<anonymous>)
    at Package.activate (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:3411343)
    at PackageManager.activatePackage (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:386057)
    at /Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:915799
    at r (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:11:910405)
    at n (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:14:1174456)
    at /Applications/Atom.app/Contents/Resources/app/static/<embedded>:14:1174610
    at Socket.<anonymous> (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:14:1173212)
    at Socket.emit (events.js:228:7)
    at Pipe.<anonymous> (net.js:664:12)

The error was thrown from the [pymakr package](https://github.com/pycom/pymakr-atom). This issue has already been reported.

Any help would be greatly appreciated. Thanks in advance.
Chris

jakobrosenberg, 2022-04-28:
Hi @RaspChris , could you try install Node and see if that fixes your issue?

Alternatively, you can also try the upcoming v2 of Pymakr for VSCode, which shouldn't have any Node dependency. https://marketplace.visualstudio.com/items?itemName=pycom.pymakr-preview

clavay, 2022-04-28:
> Hi @RaspChris , could you try install Node and see if that fixes your issue?

This works for me on Debian 11 :
> sudo apt install npm

RaspChris, 2022-04-29:
Well, sorry but no success yet (probably my lack of experience).

1) sudo apt install npm gave me : Unable to locate a Java Runtime. Please visit http://www.java.com for information on installing Java. After installing java, same issue. (I can navigate terminal a bit but not so much)

2) Pymakr 2 in VSC : improvment as pymakr could recognise the existence of the board !!! However it didn't seem to access it. I tried safe reboot to no avail
[image]

3) Pymakr 2 in Atom : same thing Pymakr see the board but cannot connect 

Now what worries me is that I upgraded the firmware of the board (before understanding that pymakr was a problem). Could it be that it went wrong ? 

Thank you.
Chris

RaspChris, 2022-04-29:
Connection timed out.
Failed to connect (Error: Port is not open).

Thank you
Chris

jakobrosenberg, 2022-04-30:
If you connect your device in Pymakr Preview and open a terminal, do you get any feedback if you spam the terminal with `CTRL + C` or `CTRL + F`?

If not, I think the issue is with your device. In that case, you can try reflashing it or contact Pycom technical support.

> sudo apt install npm gave me : Unable to locate a Java Runtime. Please visit http://www.java.com/ for information on installing Java. After installing java, same issue. (I can navigate terminal a bit but not so much)

Java is entirely different from Javascript. I'm not sure why you got that error, but `npm` is part of `node`, so I assume that's the package you meant to install.

RaspChris, 2022-04-30:
Thank you so much for answering !!!

So pymakr now opens a terminal and try connecting with the right USB address when I plug in the device. But doesn't connect.
Safe booting doesn't work... Ctrl+ C or F neither.

I will try reflashing. Thank you
Chris

jakobrosenberg, 2022-05-02:
Did you manage to get it working? 🙂

RaspChris, 2022-05-02:
No.
I tried several things none worked (no message)
When I release the reset button from the Lopy4, I get :

ets Jun 8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff8020,len:8
load:0x3fff8028,len:2128
load:0x4009fa00,len:19760
entry 0x400a05bc

I contacted support....